XenGi/teensy.pretty #2: Missing pads on Teensy++ 2.0 footprint.

josuegaleas, 2017-08-10:
After using the symbol found [here](https://github.com/XenGi/teensy_library) and generating/reading the net list, I got the folllowing errors:

[image]

The pads 13, 53, and 54 are missing from the footprint. I noticed that where 13 should be, it says 19. In other words, there are two pad 19's. Pads 53 and 54 seem to correspond to the E4 and E5 pads shown below:

[image]

TroyFletcher, 2017-09-06:
I have the same issue. Would a pull request to get the 13 updated be easier?

XenGi, 2017-12-07:
Looking into it now.

closed 2017-12-07 by XenGi