XenGi/teensy.pretty #13: Missing SMD pads on the back of Teensy 3.0, 3.1 and 3.2

jbabio, 2021-01-20:
Teensy models 3.0, 3.1 and 3.2 do have 14 pads to solder a row of 2x7 SMD pins on the back (LC doesn't). Would be nice to have those pins on the footprint to be able to route the tracks to them.

I suggest to have one footprint for the LC and another for the 3.0, 3.1 and 3.2.

https://www.pjrc.com/store/teensy32.html

blackketter, 2021-01-21:
There is a separate Teensy LC footprint, but you are right, the 3.x footprint should have the bottom pads (and remove the label for LC.). I'll take a look.

blackketter, 2021-01-21:
I've created a new footprint called Teensy30_31_32_All_Pins and updated the Teensy 3.2 symbol to include PIN numbers for all pins.  Previews:

[image]
[image]

How does this look?

jbabio, 2021-01-21:
It looks great! I'm looking forward to use them.

blackketter, 2021-01-21:
Ok, I've pushed the changes to this repo and teensy.library.  Try them out and reopen this issue if there are any problems!

closed 2021-01-21 by blackketter