XenGi/teensy.pretty #14: Teensy 3.6/3.5: Missing back SMT pads on Footprint

Qcraft, 2021-02-10:
Basically the same issue as #13

Would also be nice if the USB device pins (D+ / D-) on the back, bellow the micro b connector could be used.

blackketter, 2021-02-10:
Updated to add SMT pads to footprints (and symbols in https://github.com/XenGi/teensy_library).   There are holes for the D+/D- USB pads, but not sure what connector you'd put on the bottom of the board.  Probably will need to put a pogo pin there to make contact with the exposed pads.

Qcraft, 2021-02-12:
Thanks! however it seems like the added USB pins are both set to pin 28 which is used for the PROGRAM pin.

Also the USB pins are not on the schematic symbol.

blackketter, 2021-02-12:
Thanks, I should have reviewed it more carefully!

blackketter, 2021-02-12:
I've committed a footprint with the correct pin numbers and symbols with the missing pins.  Please let me know if you see any other issues.  Thanks!

closed 2021-02-12 by blackketter